corot/world_canvas #48: Missing warehouse_ros python import

ruffsl, 2016-11-13:
I've installed world_canvas from apt from the ROS ubuntu repo,
and I'm not sure where is this python library: `warehouse_ros` is:
https://github.com/corot/world_canvas/blob/5d48a2c9f1fea946ae50f259d86bc1999a8c98a4/world_canvas_server/src/world_canvas_server/annotations_server.py#L40

Just testing out the first launch example from the readme, I get this error:
``` terminal
$ roslaunch world_canvas_server world_canvas_server.launch debug:=true --screen
... logging to /home/ruffsl/.ros/log/35a221f2-a7a7-11e6-82a1-2c56dc38be90/roslaunch-cog-1-4257.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cog-1:40164/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.5
 * /world_canvas_server/auto_save_map: False
 * /world_canvas_server/start_map_manager: True

NODES
  /
    world_canvas_server (world_canvas_server/world_canvas_server)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[world_canvas_server-1]: started with pid [4275]
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/world_canvas_server/world_canvas_server", line 6, in <module>
    import world_canvas_server
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/world_canvas_server/__init__.py", line 1, in <module>
    from .annotations_server import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/world_canvas_server/annotations_server.py", line 40, in <module>
    import warehouse_ros as wr
ImportError: No module named warehouse_ros
[world_canvas_server-1] process has died [pid 4275, exit code 1, cmd /opt/ros/kinetic/lib/world_canvas_server/world_canvas_server true __name:=world_canvas_server __log:=/home/ruffsl/.ros/log/35a221f2-a7a7-11e6-82a1-2c56dc38be90/world_canvas_server-1.log].
log file: /home/ruffsl/.ros/log/35a221f2-a7a7-11e6-82a1-2c56dc38be90/world_canvas_server-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

wassimhariri93, 2016-11-18:
I am having exactly the same issue, was this solved with you?

wassimhariri93, 2016-11-18:
actually you need to download the package warehouse_ros from here https://github.com/ros-planning/warehouse_ros so that it can compile, but i am having now another error, instead of ImportError i am getting the following: AttributeError: 'module' object has no attribute 'MessageCollection'

ruffsl, 2016-11-20:
Hmm. @wassimhariri93 , I'll give your suggestion a try.

jimjarry, 2018-03-14:
I am having exactly the same issue, was this solved with you?

Au-ChiKin, 2019-08-13:
I have the same issue, could you please share how you solved it?

NOT-old-wang, 2020-03-23:
I have the same issue, could you please share how you solved it?

zhaokexue, 2020-04-17:
I have the same issue, could you please share how you solved it?

zhaokexue, 2020-04-17:
I have the same issue, could you please share how you solved it?@ruffsl @wassimhariri93 @Au-ChiKin @NOT-old-wang

5730289021-NN, 2020-05-20:
I've fixed this problem by importing warehouse_ros_mongo instead of warehouse_ros

zhaokexue, 2020-07-02:
> I've fixed this problem by importing warehouse_ros_mongo instead of warehouse_ros

Thank you,i solved the problem.ref:https://blog.csdn.net/zhao_ke_xue/article/details/107081543

zhujinwen0924, 2021-06-04:
> > I've fixed this problem by importing warehouse_ros_mongo instead of warehouse_ros
> 
> Thank you,i solved the problem.ref:https://blog.csdn.net/zhao_ke_xue/article/details/107081543

hello, i did the method as you said,but there is another problem:
process[world_canvas_server-1]: started with pid [7566]
Traceback (most recent call last):
  File "/home/leoh/Desktop/tmp/src/world_canvas/world_canvas_server/scripts/world_canvas_server", line 6, in <module>
    import world_canvas_server
  File "/home/leoh/Desktop/tmp/devel/lib/python2.7/dist-packages/world_canvas_server/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/leoh/Desktop/tmp/src/world_canvas/world_canvas_server/src/world_canvas_server/annotations_server.py", line 40, in <module>
    import warehouse_ros_mongo as wr
ImportError: No module named warehouse_ros_mongo
[world_canvas_server-1] process has died [pid 7566, exit code 1, cmd /home/leoh/Desktop/tmp/src/world_canvas/world_canvas_server/scripts/world_canvas_server true __name:=world_canvas_server __log:=/home/leoh/.ros/log/9c0ec688-c4ef-11eb-89b9-b42e99c7c3d1/world_canvas_server-1.log].
log file: /home/leoh/.ros/log/9c0ec688-c4ef-11eb-89b9-b42e99c7c3d1/world_canvas_server-1*.log

warehouse_ros_mongo can not be found,i don't know why,cao you help me ?thanks.

l376571926, 2022-09-30:
> 

用这个代替

https://github.com/TheBrewCrew/warehouse_ros_mongo.git